jascoproducts/firmware #12: [Firmware Request]: Enbrighten-GE - ZW4005 - 14291

raman325, 2022-06-10:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4005

### Product Number

14291

georgejung, 2022-06-10:
I have a bunch of these as well and would appreciate a firmware update. Thank you for doing this!

markaggar, 2022-06-11:
Ditto - I have a bunch of these and hoping the latest firmware allows for scenes (e.g. hold and multi-press) and faster response times.

magno101, 2022-06-11:
I've got ~20 of these, would love to get an update.

niemyjski, 2022-06-11:
I have 15 of these and would love an update

raman325, 2022-06-12:
@jascoproducts i would suggest as you assign these to yourselves that you add a comment instructing people to thumbs up the original post to express interest and then lock the thread. It'll be much easier for you gauge interest that way. You may also be able to do this automatically via GitHub workflows, not sure

jascoproducts, 2022-06-14:
@raman325 @georgejung @magno101 @niemyjski Thank you for your interest. This is currently slated for this fridays release.

@markaggar Keep an eye out in the coming weeks. We will be working on a method for Feature updates. I provided more explanation from a similar question/comment in the following link if you are interested. https://github.com/jascoproducts/firmware/issues/5#issuecomment-1155317433

jascoproducts, 2022-06-17:
Firmware provided in this week's Friday release.

closed 2022-06-17 by jascoproducts