jascoproducts/firmware #14: [Firmware Request]: GE - ZW4002 - 14287

Cossid, 2022-06-10:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4002

### Product Number

14287

Spartan-II-117, 2022-06-10:
Seconded

s7726, 2022-06-11:
With a fix for the issue where it locks up after a few uses on a secure leave connection, requiring a hard reset.

s7726, 2022-06-15:
@jascoproducts understood, sorry for the confusion, does initial release on all of these mean, initial release to the community? Or initial release to production?

Should I suspect this resolves the issues I'm seeing?

jascoproducts, 2022-06-17:
> @jascoproducts understood, sorry for the confusion, does initial release on all of these mean, initial release to the community? Or initial release to production?
> 
> Should I suspect this resolves the issues I'm seeing?

Depending on your device's date code, we issued a firmware update for this device that fixed some issues dealing with reset and exclusion behavior. See the complete revision history [here.](https://github.com/jascoproducts/firmware/blob/69a04fa25ee13768fce78076bf2d53594b96b97a/zwave/Enbrighten-GE/ZW4002/14287%20-%20In-Wall%20Smart%20Fan%20Speed%20Control/14287_Revision_History.csv)

If a firmware update does not resolve your issue, please create a new Bug Report using the existing form.

Closing this issue with firmware provided on this week's Friday release.

closed 2022-06-17 by jascoproducts