jascoproducts/firmware #5: [Firmware Request]: GE - ZW4002 - 18756

crxporter, 2022-06-10:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4002

### Product Number

18756

mastakebob, 2022-06-10:
Bump. 

A firmware that lets you turn off the indicator LED when the fan is in a steady state. The 3up-1down press combo works only works a few hours before reverting to always on. Annoying to have a blue led shining all night.

crxporter, 2022-06-10:
I'd love to have the LED on these turn off when the fan is off. I don't know if newer versions of the device can do that? 

I mean, has Jasco *made* a firmware with that feature? 

@jascoproducts is this a place to do feature requests? Should we make new issue(s) for that? Is that in the scope of this project?

marcus-j-davies, 2022-06-10:
@mastakebob

i'm not connected to Jasco in any form - but they will be releasing FW files based on most requested.
IMO - its better to submit a FW File request (using the template). - then adding a 2nd comment to it, for any additional requests.

the more a FW is identified in the list - the more chance of it being released/updated

crxporter, 2022-06-10:
That'll be a lot of requests, Marcus... could be good to "thumbs up" or add "plus one" comment to existing issues?

marcus-j-davies, 2022-06-10:
Good shout @crxporter, my row-by-row analysis kicking in 😉

jascoproducts, 2022-06-14:
@crxporter Appreciate the interaction. We have questions and responses based on the above, definitely apologize for the length, but want to make sure everything is addressed.

1. Your ZW4002. the number you identified as Product Number seems like its probably the PIN number under the QR code on the front? is that correct or did you get it from another location?

2. Does the Logo on the top left say GE or Enbrighten specifically

These 2 questions should help us identify specifically which model you have. If your model has the QR Code sticker in the bottom right of the metal yoke, and is listed as Enbrighten in the top left, the Model is 55258 and we have listed all firmware currently available for this model.

3. Regarding Feature requests, they are absolutely in the scope of this. We will be providing a method for the community to request updates, but we arent quite ready yet. Our primary focus is verifying and getting all of our current product firmware first, then we we can start handling new features. 

@marcus-j-davies @mastakebob We haven't identified exactly how we want to implement feature updates, but as we see conversations like what you are suggesting in these comments we are definitely taking note. 

Just to share a little bit more: There are many pieces at play we have to work through with feature requests, hardware limitations/older configurations and what each model is capable of, so we will have to handle each on a case by case basis. We are going to figure it out though, and bring everyone along.

jascoproducts, 2022-06-14:
@crxporter Thanks for the additional information. As long as this information is directly reporting from your device then we can use it as well. (its actually the best way to identify the product) Its hit or miss what each Z-Wave controller will allow to report or how it will report the data (hex, decimal, or not at all), so the physical device is the most definitive way to identify which product it is. 

We are trying to update requests as well with examples and will continue to try and hit all the different methods to identify and better help explain when different people come to submit requests, etc. If the screenshot you provided is directly reporting from your device, then this is SKU: 14287. This is the Non-S2/SmartStart version of the Fan Speed Control originally released. That model is slated for release this Friday for all the current available firmware files.

jascoproducts, 2022-06-17:
@crxporter From our conversation above, it appears the SKU you're referencing is 14287. If you feel like this is incorrect, please reopen another firmware request.

For now, we are closing this issue as 14287 firmware was just released in this week's Friday update.

Thanks for all the interaction.

closed 2022-06-17 by jascoproducts