jascoproducts/firmware #19: [Firmware Request]: <Brand> - <ZW Number> - <Prod Number>

yurdle, 2022-06-11:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4002

### Product Number

55258

yurdle, 2022-06-11:
Hoping for a firmware update with a fix to the status LED issue documented at  https://community.smartthings.com/t/enbrighten-z-wave-plus-smart-fan-control-55258-led-status-light/239528

TL;DR: This models status LED should be on while the load is off, its advertised in the manual, but it currently only flashes for 10 seconds when the fan is turning on, remaining otherwise off. This model should match prior model behavior(18756).

cbarth3, 2022-06-11:
Thank you for bringing this up as this is what I got previously from the support group. 
Thank you for contacting Jasco Products Company. Unfortunately, this model does not have the ability to change the LED indicator manually or in the parameters.
Now I have 1 out of close to 20 that doesn't have the light on cause my previous one failed just out of warranty.

jascoproducts, 2022-06-14:
@yurdle @cbarth3 Thanks for your suggestions. At this time our primary focus is getting all current firmware released versions loaded. We are releasing updates every Friday with as many as possible. We will be offering a method for product suggestions but do not have everything worked out how to best accomplish that. I provided some additional information on a similar comment if you are interested https://github.com/jascoproducts/firmware/issues/5#issuecomment-1155317433

closed 2022-06-14 by jascoproducts