jascoproducts/firmware #21: [Firmware Request]: <Brand> - <ZW Number> - <Prod Number>

Pau1ey, 2022-06-11:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW3003

### Product Number

12724

ConnectedHome-Jasco, 2022-06-13:
@Pau1ey Thank you for your request. Unfortunately Z-Wave products did not support Over the Air Updates until Z-Wave Plus (500 Series) models. the model you have provided is an earlier Z-Wave Classic (Z-Wave 300 Series or before) product that cannot be updated.

bawitdaba, 2022-06-14:
I have several of these older dimmer switches, they are not Z-Wave Plus to my knowledge so I don't think they can be OTA updated.

Is it technically possible for us to manually update the firmware of these switches (assuming a newer version is available) via a JTAG/serial interface or something?

I really want double tap support for them which I believe requires Z-Wave Associations to be supported and just wasn't that far back :(

jascoproducts, 2022-08-03:
Updated title to reflect Z-Wave JS database fix.