jascoproducts/firmware #23: Your Revision History files should probably be csv files

NatoNathan, 2022-06-11:
You are use]ing Excel for your Revision history, these should probably be csv files, so that GitHub can render them as a table. This will become beneficial as you continue to add updates.

crxporter, 2022-06-11:
I fully agree with your comment - I downloaded and opened several of the csv files yesterday and was slightly annoyed.

Then I noticed that the revision history is *also* available in the "readme" file right next to each downloadable file. Would be awesome if both were file types that play nice with GH.

Example:
[image]

niemyjski, 2022-06-11:
Also, I was looking over the notes for the switches I have that you've provided. The release notes say something like initial release, but my firmware is much older. Would be nice to see a change log for all versions of the firmware.

closed 2022-06-14 by jascoproducts