jascoproducts/firmware #30: Source code availability?

qwertychouskie, 2022-06-12:
I know this is probably outside of the scope of the initial release, and will likely need additional signoffs from the legal department, but is there any chance to eventually have the source code itself released?

There are a number of benefits of this both to the consumers and Jasco themselves:

 - Passionate customers can add features and fix issues themselves, and these improvements can make their way to official releases, this benefiting all users of the product
 - Potential customers that are passionate about open firmware (for reasons above) are more likely to pick Jasco products over competitors
 - More eyes on the code means that any potential security exploits can be found and patched much sooner

Thanks for everything that's been done so far, and hoping for even more openness in the future!

WillCodeForCats, 2022-06-12:
Does any z-wave manufacturer actually do that?

marcus-j-davies, 2022-06-13:
Whilst I agree with the points above, I can't imagine this will be happening anytime soon.
I would imagine high levels of IP is involved with it all, but then again, you never know.

OmgImAlexis, 2022-06-14:
@WillCodeForCats this could be one of the first. Would be nice to see a company really do a 180 on their stance.

jascoproducts, 2022-06-16:
@qwertychouskie Internally this has been brought up to discuss from other suggestions. There are indeed many factors to have to contend with like @marcus-j-davies has mentioned that would need to be reviewed and definitely legal ramifications that would have to be figured out for various reasons. Currently it isn't on the roadmap for this project, but it will likely be reviewed more in depth in the future so we can offer a definitive response with more information.