jascoproducts/firmware #44: Question: Date Codes? (ZW4006/26931)

savageautomate, 2022-06-14:
Can the latest firmware version **[5.39](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4006/26931%20-%20In-Wall%20Smart%20Motion%20Switch/5.39)** be installed on  **[ZW4006/26931](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4006)** (_Enbrighten-GE In-Wall Smart Motion Switch_) with older date codes prior to `1836` ?

I was unclear whether the date codes listed in the [README](https://github.com/jascoproducts/firmware/blob/main/zwave/Enbrighten-GE/ZW4006/26931%20-%20In-Wall%20Smart%20Motion%20Switch/5.39/README.txt) files also represents a hardware change that would limit the firmware version to specific hardware revisions?

Just trying to avoid bricking any devices.

Thanks, Robert

jascoproducts, 2022-06-16:
@savageautomate Thank you for the question. We apologize for the delayed response, but we wanted to verify some specific hardware information before we responded with a definitive answer.

Short answer is: yes, you are fine. 

Longer explanation is: as long as you have identified the product correctly using the "Identifying your Product" section of this repository on the GitHub , you can use any of the files we have released for that specific Model "SKU" number (26931) of Hardware version (ZW4006). We wouldn't recommend to downgrade to an earlier version without having a specific reason, but if you do, it won't cause any issue with the hardware.

savageautomate, 2022-06-20:
Thank you.  I have upgraded these older units to the latest 5.39 firmware with no issues.

closed 2022-06-20 by savageautomate