jascoproducts/firmware #50: [Bug Report]: Enbrighten/GE - ZW4002 - 55258 - 5.50

yurdle, 2022-06-14:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW4002

### Product Number

55258

### FW Version

5.50

### Bug Report Details

Jasco [manual for ZW4002/55258](https://byjasco.com/mwdownloads/download/link/id/2563) calls out that the blue status LED will be illuminated in the dark to help you find the switch, but 55258 does not leave the LED on when the load is switched off, and there does not exist a configuration value to change the behavior.
[image]

This is reported here as well: https://community.smartthings.com/t/enbrighten-z-wave-plus-smart-fan-control-55258-led-status-light/239528

This is not a feature request, this should be considered a bug/defect as the manual calls it out as already being included and previous generations of ZW4002 worked as intended. 

Further, I personally have 5 of these installed and all behave the same way. I had one replaced under warranty and the replacement exhibit the same behavior, so I am confident this is not a hardware issue, but a firmware bug. 

### How To Reproduce

Install a new Enbrighten ZW4002/55258. 
Power on the circuit. 
Find that the status LED remains off. 
Turn the fan on, status LED flashes to indicate speed, then stops blinking, remaining off.
Turn the fan off, status LED remains off.

cbarth3, 2022-07-04:
This is a big issue for me as every other switch in my home (20+) has the ability to have the light on when the switch is off. This needs to be added back in for this newest version which is the only version available.

Rajamahal, 2022-07-06:
+1 to having a bunch of these switches and no configuration option for the LED. Request firmware fix

nathan-curtis, 2022-08-31:
+1, Has this been fixed?

Just discovered this in the new Enbrighten version switches, I had an older one fail and had to replace it.  I was excited to get ZWave Plus support.  I do understand that this issue **seems** minor - but this is an issue big enough to not choose this device and instead go to another vendor.  In that case I can explain the difference in functionality.  The end user perception now is the fancy 'new, better' fan controller 'Doesn't work right.'

jascoproducts, 2022-09-01:
Hi all.

Can't make any promises on a deadline as we haven't finished an initial version yet for testing, but development for a corrective firmware is well underway. It will be a Z-Wave Alliance-certified full release.

Thanks for your patience and sorry for the delay.

cbarth3, 2022-09-12:
> Hi all.
> 
> Can't make any promises on a deadline as we haven't finished an initial version yet for testing, but development for a corrective firmware is well underway. It will be a Z-Wave Alliance-certified full release.
> 
> Thanks for your patience and sorry for the delay.

Just wanted to check and see if there are any updates?

jascoproducts, 2022-09-12:
> Just wanted to check and see if there are any updates?

Initial firmware development is complete.

Should finish testing and be off for certification tomorrow!

superdupe, 2022-09-22:
Can I add a +1 to this request?  Looking forward to a fix for this led issue... :)

cbarth3, 2022-10-01:
> Initial firmware development is complete.
>
> Should finish testing and be off for certification tomorrow!

Any more updates on possible release?

superdupe, 2022-10-04:
I wonder what happened.  They were responding and everything was good up until about three weeks ago, and all of a sudden, radio silence... Hope whoever was the person responding on here is ok...

jascoproducts, 2022-10-04:
@cbarth3 @superdupe 

Sorry we missed the last response to this thread.

The corrective version for this issue is still in certification which we didn't think would take so long. I am going to release a temporary beta which will be up for download indefinitely until the final version has finished certification. You will be able to use it immediately.

superdupe, 2022-10-04:
Cool.   Thank you... :)

superdupe, 2022-10-05:
Were you able to post the beta firmware?  I don’t see it anywhere.  Maybe I missed it?

jascoproducts, 2022-10-05:
@superdupe should be ready tomorrow. Sorry for the delay. Will update here once it's up.

jascoproducts, 2022-10-07:
Hi guys. We are having problems getting the beta version to update to the device properly during testing. We will try to get the version out by Monday at the latest.

jascoproducts, 2022-10-10:
[Corrective beta firmware now available!](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4002/55258%20-%20In-Wall%20Fan%20Speed%20Control%2C%20500S/0.51%20(Latest%2C%20Beta))

jascoproducts, 2022-10-11:
Closing issue: Corrective Beta firmware released. Originally this firmware version was scheduled to be officially released to mass production, however after review, no further production is scheduled for this model and product will be discontinued while we are moving to new hardware platform and updated Z-Wave 700 series.

cbarth3, 2022-10-11:
> Closing issue: Corrective Beta firmware released. Originally this firmware version was scheduled to be officially released to mass production, however after review, no further production is [...]

Will this be pushed/available through home assistant like other versions?

jascoproducts, 2022-10-11:
@cbarth3 Yes, we will be opening a pull request with them shortly to allow for updates through the HA service.

Rajamahal, 2022-10-17:
Anyone here get this firmware to work on Home Assistant? It keeps failing for me.

superdupe, 2022-10-17:
I don't use HA, I use Hubitat, but I do all my firmware updates by adding a USB stick with PC Controller to the hub.  Its easier to do some functions with PC Controller than with the Hubitat hub.  That being said, I was able to update all 6 of my devices to this firmware and can report it works!  The blue LED is on with the switch off...

My only issue, and not a big deal, is this firmware is named .51 instead of 5.51... Maybe because it is beta and will not ever be released as an actual update to 5.50?

patchrick84, 2022-12-18:
Did this work for anyone else?  I used Z-Wave JS to upload it to the switch and it seems to succeed but I never get any new configuration options in the node's settings.

[image]

kyleguillott, 2022-12-18:
Nope. Broke mine even more. Now my fan only works at 100%. Anything less than that and it turn off after a second or two.

yurdle, 2022-12-18:
@patchrick84 Yes, the beta release resolved my issues. Note that this model never had the config parameter documented, so zwaveJS doesn’t expose it.
Use “Custom configuration” to manually set config parameter 3 to a value of 2 to set the LED on when the fan is off, matching the documentation for the older controller, 14314: http://manuals-backend.z-wave.info/make.php?lang=en&sku=14314&cert=ZC10-15060022&type=mini

Rajamahal, 2022-12-18:
@patchrick84 Yes eventually but it took many update attempts per switch through home assistant to get it to 100% completion without freezing up and having to remove node, add, try again.

meluvalli, 2022-12-20:
I am also running HomeAssistant.  I am not sure how to configure this manually.  I updated to the firmware 0.51 successfully, and HA shows I am on that firmware for the device, but I can't change the parameter.  I tried running the service manually with the following:
<code>
service: zwave_js.set_config_parameter
data:
  parameter: "3"
  value: "2"
target:
  entity_id: fan.in_wall_3_speed_fan_control
</code>
However, it just fails.  And doesn't change anything.  Any suggestions?

Rajamahal, 2022-12-20:
> I am also running HomeAssistant. I am not sure how to configure this manually. I updated to the firmware 0.51 successfully, and HA shows I am on that firmware for the device, but I can't change the parameter. I tried running the service manually with the following:
> ` service: zwave_js.set_config_parameter data: parameter: "3" value: "2" target: entity_id: fan.in_wall_3_speed_fan_control `
> However, it just fails. And doesn't change anything. Any suggestions?

[image]

jascoproducts, 2022-12-20:
Reopening this as an ongoing discussion and to make room for a new pin.

meluvalli, 2022-12-21:
> > I am also running HomeAssistant. I am not sure how to configure this manually. I updated to the firmware 0.51 successfully, and HA shows I am on that firmware for the device, but I can't change the parameter. I tried running the service manually with the following:
> > `service: zwave_js.set_config_parameter data: parameter: "3" value: "2" target: entity_id: fan.in_wall_3_speed_fan_control`
> > However, it just fails. And doesn't change anything. Any suggestions?
> 
> [image] @meluvalli

Thank you!  I installed and configured Z-Wave JS UI and this worked.  I am not sure why I couldn't do it without the UI with the commands I sent before, but this did the trick!  Thanks again!

patchrick84, 2022-12-22:
> [image] @meluvalli

This also worked for me, although I needed to use a value of 1 instead of 2 for some reason.  Thank you for the hint!

jascoproducts, 2023-12-04:
Cleaning up issues - closing this issue as resolved.

clarkja5, 2025-10-19:
I forgot to mention I had to do a manual update with the firmware from the GitHub repository it wouldn't work using ota updates.

clarkja5, 2025-10-20:
I have both because at first I thought maybe the new firmware was the problem, but I think I used the one in the zip. I'm just sitting down to really look at it since that update. I am running Zwave JS Ui running in HA. It is currently showing FW: v0.51 but it is giving me the option to upgrade to 5.51. I think I'm going to let it try and see what happens.

steverep, 2025-10-20:
> I just went into devices in HA and the firmware listed there is v5.51

This doesn't match your previous comment, and it's not what I see.

clarkja5, 2025-10-21:
I have taken another look and I see where I saw v5.51. It shows v0.51 in the Zwave JS UI node list and in HA devices under device info, but if I scroll down to the Configuration box that is where it says Zwave firmware is 5.51

nmpu, 2025-11-01:
The versioning issue has been resolved. Jasco had simply renamed the file without updating the internal version.

@jascoproducts Let's suppose parameter 40 is set to 1: _Press the paddle up **or** down **once** to cycle between fan speeds_.

I understand how tap up could work. In fact, I already have this implemented using the raw button events. I have it coded so that tapping up _while already on_ will cycle low->medium->high->low. I also have it coded so that tapping down _when already off_ will start on low. In all other cases, the default behaviors are maintained-- tapping down is instant off and tapping up resumes at the previous value.

 How is your tap down supposed to work? If the fan is on high, do we need to tap down 3 times to turn off?