jascoproducts/firmware #54: [Bug Report]: Downloading all firmware files

alexruffell, 2022-06-19:
### What brand is the bug report for?

Jasco

### ZW Number

All

### Product Number

All

### FW Version

All

### Bug Report Details

There is no easy/direct way to download the files. When I click on the file name it opens the file showing me the contents. If I right-click and pick "Save link as..." I get a file but it is actually the HTML of the page. I noticed the file I got was unusually large but never saw a Jasco firmware file before so did not think much of it. After tons of failed attempts I took a peek inside and saw it was HTML, not hex.

I went back to your files, opened one I needed, hit "Raw", copied the contents and saved it to a file with .hex extension. I uploaded that file and it worked but it is prone to making mistakes and causing devices to fail.

Can you make it easier to download the files?

### How To Reproduce

See above

Holladiewal, 2022-06-19:
You should be able to download the raw file by using the "Save link as..." functionality on the Raw button itself, as this is just a link to the raw file.
Besides directly linking in the folder-local readme, Jasco will probably not be able to change much of Github's behaviour.

alexruffell, 2022-06-19:
@Holladiewal - That worked perfectly - thank you.

I am quite sure I've seen other github repositories that link to the files for easy download but your solution works. Maybe they could put a note somewhere to make it easier for the github novices like myself :)

crxporter, 2022-06-19:
I just downloaded the entire repo to my computer then pulled up the files I need. There's a button to download the whole thing as a zip on the main page of the repo.

alexruffell, 2022-06-19:
@crxporter Ah! Code >>> Download Zip... missed that too!

jascoproducts, 2022-06-20:
Hello, and thanks for the detailed information/suggestion. We agree that it is a bit counterintuitive that there is no way to download folders individually.

We recommend using [DownGit](https://downgit.github.io/#/home) or [similar](https://stackoverflow.com/questions/7106012/download-a-single-folder-or-directory-from-a-github-repo) to solve this.

First, navigate to the contents of your desired firmware version and copy the URL:
[image]

Paste the URL into DownGit or your desired downloader and hit the Download button:
[image]

We at Jasco don't have any plans to develop, repository-based or otherwise, alternatives to the services mentioned above, but we would very much appreciate ideas. We will leave this issue open and postpone adding the above steps to the repository readme to promote discussion and consideration of a potentially more elegant solution to the problem.

jascoproducts, 2022-06-20:
How would you all feel about us reuploading the firmware files as .zip archives? Github has these available for immediate download, individually. Thoughts?
[image]

crxporter, 2022-06-20:
Would it be too much to *add* the zip to the current available downloads? This way we can see the readme, raw file, and easy-to-download zip file.

PS: 900 quintillion bytes? That's a big firmware.

alexruffell, 2022-06-20:
@jascoproducts I vote for @crxporter 's suggestion if at all possible.

marcus-j-davies, 2022-06-21:
@jascoproducts, @crxporter 

One solution, is to add a **README.md** in the root folder of each brand.
The Read Me can then use GitDown's link function, so the user clicks the link, and it will start to download that brands FW collection as a zip file.

This way, a zip folder being created each time, wont be needed, but the link will alley download the current offerings.
also allows some contextt/extra info being added to address certain Brand specific things

Example:

https://github.com/marcus-j-davies/firmware/tree/main/zwave/Honeywell

[image]

anyway, just my 2c.

crxporter, 2022-06-21:
@marcus-j-davies I think there's not much sense in downloading a *whole brand*. If you want everything just download the whole project. If you want a single FW version of a single device (most likely what people want) then a zip of just the one firmware to download would be great.

marcus-j-davies, 2022-06-21:
> If you want a single FW version of a single device (most likely what people want) then a zip of just the one firmware to download would be great.

That's what I'm trying to get at 😅 
The zipped brand, is as close to a single FW file you will get.

Most users invest in one brand or another?
Each brand folder, will have a download link (via the Readme as above) - providing them with only that brands FW files as a zipped package

The alternative is to download the entire repo of course using the Github built in methods.
but if the aim is to not have to download the entire repo - then my suggestion above, is (currently) the only way, at least because of GitHub's handling of certain files.

alexruffell, 2022-06-21:
@marcus-j-davies @crxporter I like the idea, but I agree that downloading the entire brand might be a bit much especially as it grows in models and firmware versions. I downloaded the entire archive and there was lots of clicking around to find the file.

Aren't there readme files at lower levels too? I see the benefit to setting something up that doesn't require extra maintenance work each time a firmware file is added. Maybe a link in the readme for each model might be a way in between. You end up with all the old revisions but I doubt those will be many for each model and anyway the files are tiny.

marcus-j-davies, 2022-06-21:
> I downloaded the entire archive and there was lots of clicking around to find the file.

This is exactly my point!

I think it depends on @jascoproducts and their willingness to create individual zipped files (per FW)

options currently are

 - Download entire repo
 - A zipped brand (using my suggestion above)
 - @jascoproducts zipping up all FW files individually

alexruffell, 2022-06-21:
@marcus-j-davies Can a readme be added at this level:

[image]

or

[image]

If so, wouldn't it be possible to allow downloading all firmware versions for a single model?

marcus-j-davies, 2022-06-21:
Yes it can.

any folder that has a README.md will be rendered, when you navigate to that folder

Example

https://github.com/marcus-j-davies/firmware/tree/main/zwave/Honeywell/ZW3010/39351%20-%20In-Wall%20Smart%20Paddle%20Dimmer%20(QFSW)

marcus-j-davies, 2022-06-21:
This approach (at whatever level  - the further down, the better) coupled with the GitDown link feature, should make it fairly intuitive.

crxporter, 2022-06-21:
@marcus-j-davies says:
> zipping up all FW files/devices individually < -- obviously better, but requires more time

Silly Marcus, you're the automation master. Surely you (or someone at Jasco) could make quick work of automagically making and uploading these files. GitHub surely has APIs and Bots that would knock that out in no time.

marcus-j-davies, 2022-06-21:
Create an action to trigger on `push`

 - Zip up any file ending with one of the known FW extensions.
 - Get the action to upload these files to the repo.
 - Delete the original file 

🤓

alexruffell, 2022-06-21:
@jascoproducts I would only really need/want the latest revision. Going back to older ones may be necessary if there is a show-stopper bug introduced in a newer version. The assumption is that creating a zip file for every single firmware is excessive burden but if it is not, then it would be great to have.

It is my understanding that the tool being discussed in the thread for zip and provide for download the file(s) automatically so placing the link as close as possible to the smallest number of files would reduce your work to provide and maintain zip files, and it would reduce the size of what the user needs to download. At this point in time, I would dare to say that those downloading zwave (and hopefully zigbee in the near future... hint hint) firmware files are more technical than the average person so ending up with a few extra files should not be too much of an issue if it makes it more feasible for you to provide the zip archive for easier download.

marcus-j-davies, 2022-06-21:
@jascoproducts 

To repeat the comment by @alexruffell 
my suggestion here https://github.com/jascoproducts/firmware/issues/54#issuecomment-1161326262

is to create a **README.md** at the smallest possible stopping point (or at the largest resolution you are willing to go for :wink:)

And the read me containing a link to its folder.

example: 
This link https://minhaskamal.github.io/DownGit/#/home?url=https://github.com/jascoproducts/firmware/tree/main/zwave/Honeywell

will download the zip folder containing files in Honeywell

crxporter, 2022-06-21:
Plot twist, if anyone didn't know: Marcus is in
the UK and doesn't have any Jasco stuff 😂 

He's just a very helpful zwave-js developer.

AlCalzone, 2022-06-22:
Considering our plans to provide (semi-)automatic firmware updates in Z-Wave JS: Whatever gets decided, the non-zipped files should stay available for individual download. That is if you're not planning to host them somewhere else too.

jascoproducts, 2022-06-22:
@AlCalzone Definitely, that goes without saying. A major priority for us to allow your team and others to automate pulling from this.

@crxporter @marcus-j-davies the Zip files at the version level are a definite simple solution with revision notes, we are looking at the links options as well to give people options, just need to figure out the best level to place them out without adding a lot of clutter or levels that don't make sense or don't benefit anyone.

I think we have everything we need. A big thank you for all the suggestions and commentary around this. Look for at least the firmware version zip files on this release. If we dont have links on readme's for this release, then it will be on next weeks release.

closed 2022-06-22 by jascoproducts