jascoproducts/firmware #56: [Bug Report]: Enbrighten-GE - ZW4002 - 14287 - 5.24 - Missing configuration parameter documentation

kpine, 2022-06-19:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW4002

### Product Number

14287

### FW Version

5.24

### Bug Report Details

The [revision history](https://github.com/jascoproducts/firmware/blob/main/zwave/Enbrighten-GE/ZW4002/14287%20-%20In-Wall%20Smart%20Fan%20Speed%20Control/14287_Revision_History.csv#L10) for 5.24 mentions that a parameter 19 was added. I haven't found any information about what this is. Can you provide a specification for it (description, possible values and size)?

### How To Reproduce

N/A

kpine, 2022-06-20:
I'm guessing parameter 19 is "Alternative exclusion".

jascoproducts, 2022-06-20:
> I'm guessing parameter 19 is "Alternate exclusion".

It is indeed alternate exclusion. Planning to issue updated revision histories for applicable models where this is not described; leaving this issue open until then.

jascoproducts, 2022-06-20:
Decided to reopen for now to be consistent with our issue handling procedures Will close after documentation is updated Friday.

jascoproducts, 2022-06-27:
Parameter 19 documentation has been added. Closing this issue now.

closed 2022-06-27 by jascoproducts