jascoproducts/firmware #59: [Bug Report]: BRAND - ZW 4008 - PRODUCT NUMBER - FW VERSION

janngobble, 2022-06-26:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW4008

### Product Number

46201

### FW Version

5.54

### Bug Report Details

You posted the otz for 5.54 and the zip for 5.53. I don't know what to do w/otz cos when I click it shows a bunch of text in a text area on GitHub...no download. please upload the 5.54 ZIP file.

### How To Reproduce

see above.

tagno25, 2022-06-27:
The otz is the actual firmware file.  Right-click on raw and "Save Link As..." to save the firmware.

jascoproducts, 2022-06-27:
@tagno25 Thank you for suggesting a temporary fix.

@janngobble You can find the newly-uploaded .zip [here.](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4008/46201%20-%20In-Wall%20Smart%20Paddle%20Switch/5.54)

Thanks for reporting this issue. Fixed with latest hotfix and closing the issue.

closed 2022-06-27 by jascoproducts