jascoproducts/firmware #62: [Firmware Request]: GE/Jasco - ZW4202 - 14298

notmayo, 2022-06-30:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4202

### Product Number

14298

jascoproducts, 2022-06-30:
@notmayo thanks for the interest in our firmware! You can find 14298's firmware available for download as a .zip file [here.](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4203/14298%20-%20Plug-In%20Outdoor%20Smart%20Switch/5.53)

closed 2022-06-30 by jascoproducts

jascoproducts, 2022-06-30:
@notmayo ZW4203 should be the identifier if you have the 14298 outdoor smart switch.

Makes sense why you are seeing that though - I see in the Z-Wave JS database that 14298 is listed incorrectly as ZW4202. HA in turn queries from this database to retrieve device info. We will put this on our list to address; thanks for bringing this to our attention.

ZW4202 actually refers to an obscure 300-series product of ours that does not support OTA updates.

notmayo, 2022-06-30:
@jascoproducts Thank you for the clarification! :)