jascoproducts/firmware #69: [Firmware Request]: BRAND - ZW NUMBER - PRODUCT NUMBER

Diablo0, 2022-07-01:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW3003

### Product Number

12724

cbarth3, 2022-07-04:
Plus one on this

infamy, 2022-07-06:
I have this but its detected as Jasco/ge

jascoproducts, 2022-07-07:
@Diablo0 @cbarth3 @crxtasy12 This is a Z-Wave classic 300-series device which unfortunately does not support firmware updates.

@infamy no worries; this is indeed a GE device; Z-Wave JS currently has our Z-Wave products blanket classified under "GE/Jasco."

infamy, 2022-07-07:
It maybe worth creating stubs for these unupdatable devices, that just have a readme saying they do not support updates. (this may help reduce new tickets being created...)

jascoproducts, 2022-08-03:
Updated title to reflect Z-Wave JS database fix.