jascoproducts/firmware #70: [Firmware Request]: BRAND - ZW NUMBER - PRODUCT NUMBER

jeffkjolley, 2022-07-02:
### What brand is the request for?

Enbrighten-GE

### ZW Number

zw4201

### Product Number

14284

jeffkjolley, 2022-07-02:
I have 2 versions of ZW4201-
1 with product number 14284
1 that shows no product number. The date code is 1615. The FCC/IC IDs end with 45604 instead of 45604B like on the newer ones.

jascoproducts, 2022-07-05:
@jeffkjolley It looks like you have an older Z-Wave Classic (300 series) model and the Z-Wave Plus (500 series) model. You can easily determine this by looking at the Z-Wave Logo. If the model has Z-Wave Plus logo, that is the 14284 and we will be adding that file to the repository in this weeks update. 
The other model however if it only has a Z-Wave logo (no Plus under Z-Wave in the logo) then it is the 300 series 12720 which was developed and manufactured before the Z-Wave Alliance had standardized OTA updates. the 12720 wont be able to be updated unfortunately. If you want to take a picture of the bottom of both products where all the markings are and add them here, we can verify with full certainty that, that is the case.

jascoproducts, 2022-07-08:
[Just posted!](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW4201/14284%20-%20Plug-In%20Outdoor%20Smart%20Switch) Thanks for the suggestion. Hoping this satisfies your firmware needs.

closed 2022-07-08 by jascoproducts