jascoproducts/firmware #79: [Bug Report]: Enbrighten-GE / ZW3011 / 46204 - In-Wall Smart Toggle Dimmer / 5.54

DaDre, 2022-07-19:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW3011_Enbrighten-GE_46204_5.54.otz

### Product Number

ZW3011_Enbrighten-GE_46204_5.54.otz

### FW Version

ZW3011_Enbrighten-GE_46204_5.54.otz

### Bug Report Details

Upgrading to 5.54 made my lights flicker, downgrading back to 5.53 resolved the issue. No settings were changed between upgrade/downgrade. I use Z-Wave JS in Home Assistant for the upgrade / downgrade.

### How To Reproduce

Upgrade firmware, turn lights on, see flicker.

jascoproducts, 2022-07-21:
@DaDre Hey there, thanks for bringing this to our attention.

Nothing wrong with downgrading to 5.53; this is what we would recommend doing in this case actually.

5.54 was the exact same firmware as 5.53 with the exception of changing dimming performance to enhance compatibility with newer bulbs. This may have adversely affected performance with older bulbs as a consequence and this might be what you're seeing in your use case.

Rest assured, there were no other changes to the firmware so **you will not lose any performance or functionality if you decide to remain on version 5.53.**

closed 2022-07-21 by jascoproducts

jascoproducts, 2022-09-12:
Full list of devices that may be affected by this issue. Click on your device below to browse firmware.

- [Enbrighten-GE / ZW3010 / 46203 - In-Wall Paddle Dimmer, QFSW, 500S](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW3010/46203%20-%20In-Wall%20Paddle%20Dimmer%2C%20QFSW%2C%20500S)
- [Enbrighten-GE / ZW3011 / 14296 - In-Wall Toggle Dimmer, Lt Almd, QFSW, 500S](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW3011/14296%20-%20In-Wall%20Toggle%20Dimmer%2C%20Lt%20Almd%2C%20QFSW%2C%20500S)
- [Enbrighten-GE / ZW3011 / 46204 - In-Wall Toggle Dimmer, QFSW, 500S](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZW3011/46204%20-%20In-Wall%20Toggle%20Dimmer%2C%20QFSW%2C%20500S)
- [Jasco / ZW3011 / 46565 - In-Wall Toggle Dimmer, QFSW, 500S](https://github.com/jascoproducts/firmware/tree/main/zwave/Jasco/ZW3011/46565%20-%20In-Wall%20Toggle%20Dimmer%2C%20QFSW%2C%20500S)