jascoproducts/firmware #82: [Bug Report]: BRAND - ZW NUMBER - PRODUCT NUMBER - FW VERSION

badmuz, 2022-08-15:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW4008

### Product Number

46201

### FW Version

5.54

### Bug Report Details

Hi there.

On new 5.54 software "Double tap" function also triggers the regular "On" function.

### How To Reproduce

Set up automation with "Double Tap" action. Switch will run programmed action as well as triggers regular "On" action.

cmamatey, 2022-08-22:
I have also observed this same behavior when upgrading to FW v5.54 on the following:

- GE Enbrighten 46202 / ZW4009
- UltraPro 39348 / ZW4008

Note that the issue applies to all multi-taps, not just double-tap up.
Double and triple taps UP will both turn the load on and trigger the expected scene.
Double and triple taps DOWN will both turn the load off and trigger the expected scene.

I was able to reinstall FW v5.53 on both devices and restore the correct functionality.

jascoproducts, 2022-08-22:
@cmamatey Thanks for the feedback. We will look into these devices as well.

jascoproducts, 2022-09-01:
We have working beta firmware that includes features from 5.54 while retaining the behavior of 5.53. Still polishing a few things out; should be ready sometime next week. Will keep you all posted.

jascoproducts, 2022-09-02:
@badmuz Just posted [beta firmware for this device](https://github.com/jascoproducts/firmware/blob/1d7de3ac344fa6fd5d1ad645f5d33cbdf058ca2f/zwave/Enbrighten-GE/ZW4008/46201%20-%20In-Wall%20Paddle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4008_Enbrighten-GE_46201_0.54.zip) that should fix your issue while retaining the full functionality of v5.54.

Leaving this topic open until versions are released for models 46202 and 39348 as mentioned by @cmamatey

tuday2, 2022-09-09:
I observed same behavior and installed beta firmware and it is now behaving like it did before (double taps on/off do not trigger device on/off). Thank you!!

icemarkom, 2022-09-09:
Likewise. Several of my switches were affected by this bug (new Home Assistant upgrade feature rolled out today). Luckily, I found this great repo and this issue, and v0.54 seems to indeed resolve the problem for me.

jascoproducts, 2022-09-09:
@tuday2 @icemarkom Awesome, really happy to hear it worked for you all!

Just rolled out another update today adding beta releases of this same firmware for similar devices:

- [Honeywell ZW4008 / 39348](https://github.com/jascoproducts/firmware/blob/3b8f23e4c59ea69f184e9303929d9921d998faae/zwave/Honeywell/ZW4008/39348%20-%20In-Wall%20Paddle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4008_Honeywell_39348_0.54.zip)
- [UltraPro ZW4008 / 39348](https://github.com/jascoproducts/firmware/blob/3b8f23e4c59ea69f184e9303929d9921d998faae/zwave/UltraPro/ZW4008/39348%20-%20In-Wall%20Paddle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4008_UltraPro_39348-54890-54891_0.54.zip)
- [Honeywell ZW4009 / 39354](https://github.com/jascoproducts/firmware/blob/3b8f23e4c59ea69f184e9303929d9921d998faae/zwave/Honeywell/ZW4009/39354%20-%20In-Wall%20Toggle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4009_Honeywell_39354_0.54.zip)
- [UltraPro ZW4009 / 39354](https://github.com/jascoproducts/firmware/blob/3b8f23e4c59ea69f184e9303929d9921d998faae/zwave/UltraPro/ZW4009/39354%20-%20In-Wall%20Toggle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4009_UltraPro_39354-54912_0.54.zip)
- [Enbrighten ZW4009 / 46202](https://github.com/jascoproducts/firmware/blob/3b8f23e4c59ea69f184e9303929d9921d998faae/zwave/Enbrighten-GE/ZW4009/46202%20-%20In-Wall%20Toggle%20Switch,%20QFSW,%20500S/0.54%20(Latest,%20Beta)/ZW4009_Enbrighten-GE_46202_0.54.zip)

Need to double check a few things on my end to make sure we've covered our entire lineup of similar products. For now this issue will remain open.

badmuz, 2022-09-19:
@jascoproducts ,
Thank you so much for a quick fix. I can confirm 0.54 has fixed the issues.

icemarkom, 2022-09-23:
Will you be releasing a non-beta of this? Those of us with a recently-added Home Assistant upgrade system are still seeing upgrade available, but that would be to a broken version.

multicm, 2022-10-08:
I am looking for updates as well.

mrrobotnic, 2022-10-10:
Same here running FW 5.54 on a GE/Enbrighten 46201 / ZW4008 and have observed this behavior as well and curious to see when the finalized version of the firmware will be released. Thanks for the progress on this so far!!

nickbits1024, 2022-10-10:
I was about to make a bug for [Enbrighten ZW4009 / 46202] when I found this already was made.  This also applies to double/triple tap DOWNS.  So if the light was on and you execute a double/triple tap down, it also shuts off the light.

jascoproducts, 2022-10-10:
[We are working on a way](https://github.com/zwave-js/firmware-updates/pull/31) to introduce 0.54 as a recognized current version that will no longer prompt you to upgrade. We have no plans to release a 5.55 (a certified 0.54) as we will not be bringing in any more new inventory of the affected devices and thus will not be undergoing certification.

If HA prompts an "upgrade" to 5.54, please ignore it indefinitely as we work on the pull request.

jascoproducts, 2022-10-10:
To expand on this a bit, we are in the process of replacing the affected devices with our 700 and 800 series product line which is why we will not be replenishing and therefore certifying existing devices.

We originally introduced v5.54 without the delay as the vast majority of device owners were complaining about the delay required for Central Scene control (aka the multi tap) to function correctly, even after tuning the delay as tight as possible.

For those that prefer the functionality of Central Scene via multi-tap control, 0.54 will be your version.

mrrobotnic, 2022-10-10:
@jascoproducts Out of curiosity, since 0.54 is the best of 5.53 and 5.54 for those of us who want double tap enabled, what are the differences between 5.53 and 5.54?  I see the changelog mentions three items, but I am not well versed enough to know what they mean in practice.  Thanks!

Changes introduced in 5.54
1. Add Parameter 84 
2. Valid Value 0 – No specific behavior necessary Valid Value 1 – Reset Switch to Factory Defaults
4. Valid Value 1 – Reset Switch to Factory Defaults

mrrobotnic, 2022-10-11:
Just updated to v0.54 and it works great and brought back the double tap without also turning the light connected to the switch on and off at the same time.  

For anyone looking to update in Home Assistant running Z-Wave JS UI you can download the firmware from the link provided by @jascoproducts (quoted below), select the desired device to update, select "Advanced," under "Firmware update" click "Begin" and then select the .otz file that you downloaded, click OK, and it will start the update.  When it is done, you will see a notification in Home Assistant that there is an update to a newer version of the firmware, but you can skip the update to hide that notification.

> @badmuz Just posted [beta firmware for this [...]

jascoproducts, 2022-10-11:
> @jascoproducts Out of curiosity, since 0.54 is the best of 5.53 and 5.54 for those of us who want double tap enabled, what are the differences between 5.53 and 5.54? I see the changelog mentions three items, but I am not well versed enough to know what they mean in practice. Thanks!
> 
> Changes introduced in 5.54
> 
> 1. Add Parameter 84
> 2. Valid Value 0 – No specific behavior necessary Valid Value 1 – Reset Switch to Factory Defaults
> 3. Valid Value 1 – Reset Switch to Factory Defaults

@mrrobotnic 5.54 The 3 items you are referencing are really just the same piece of information. We introduced Parameter 84 that will cause a remote reset to factory default settings. Z-Wave does not natively support remotely telling an end device to leave the network and reset itself to factory defaults. This parameter will specifically do this. The default value is always 0. If you were to set it to 1, it would instantly reset itself, leave the network, and reset the parameter value to 0 so you can add it back the network or to another network.

jascoproducts, 2022-10-11:
Closing this thread: Beta releases have been uploaded for the 6 affected In-Wall products on all brands released (see below)

> @tuday2 @icemarkom Awesome, really happy to hear it worked for you all!
>
> Just rolled out another update today adding beta releases of this same firmware for similar devices:
>
> * [Honeywell ZW4008 / [...]

closed 2022-10-11 by jascoproducts

whoisdecoy, 2023-03-26:
first of all, thank you! if someone else stumbles upon this and gets this error when trying to update using zwavejs in home assistant, change the firmware extension from .zip to .bin.

`Z-Wave error 1506: Could not detect firmware format zw4008`

jascoproducts, 2023-03-27:
@whoisdecoy Could you perform a Manufacturer_Specific_Get for us on the device in question and attach the data it returns? Here's an example:

[image]

BrandtWoolf, 2025-07-11:
@jascoproducts I am experiencing this same behavior with `Jasco / In-Wall Toggle Switch, QFSW, 500S
 46563 / ZW4009` on firmware `5.54`. I am not seeing this model referenced here, nor another issue having the same problem. 

Is this known and/or is there a fix for this?

nmpu, 2026-09-30:
This was not a bug. It's a deliberate breaking change. Power users look for Central Scene support which is required for certification. Jasco then releases an update to neuter Central Scene. This is because the majority of basic users are frustrated by the delay required to detect multiple taps. In fact, if you don't understand how the switch works, you could (rapidly) tap endlessly and the switch would never change state. This must result in complaints and returns. The solution would be to introduce a configuration parameter or button sequence to switch modes.