jascoproducts/firmware #84: [Bug Report]: Enbrighten-GE - ZW4005 - 14291 - v5.24

tagno25, 2022-08-22:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW4005

### Product Number

14291

### FW Version

v5.24

### Bug Report Details

Double tap (commandClass 32) on add-on switched does not function, only single tap registers.

### How To Reproduce

Attempt to double tap, but singe tap recorded.

cmamatey, 2022-08-22:
I also have this issue.
I just installed an old ZW4005 14291 switch in a new location which has add-on switches.
Double-taps work correctly on the main switch, but attempting to double-tap any of the add-on switches only registers as single-taps.
I am also currently on FW v5.24, but I believe the same behavior was present on v5.22.
I suspect this could be a hardware limitation, but I would greatly appreciate a fix via firmware if possible.

jascoproducts, 2022-09-01:
@tagno25 Thanks for the report. After some investigation, it looks like this device does not support the Z-Wave Central Scene command class which allows for a double tap signal to send and register via addon switch.

@cmamatey this is indeed a hardware limitation. Sorry about that.

closed 2022-09-01 by jascoproducts