jascoproducts/firmware #95: Zigbee firmware ETA?

rolf-tx, 2022-10-07:
Can we get an ETA for Zigbee firmware release?

jasondefuria, 2022-10-07:
I was literally going to tweet at them when I had a chance about this. It says "coming soon" all summer.

jascoproducts, 2022-10-10:
Hi there. Sorry for the delay on this topic; I know it has been a while since our last Zigbee discussion. We have been working with HomeAssistant on our Z-Wave products, both past and present, to ensure the best possible user experience on that end.

While we will be releasing all Zigbee firmware, it's worth noting that, at this time, each device only has one release available (that of which should already be installed on your Zigbee device(s). There have not been any necessary updates, features or otherwise, that have prompted us to issue new versions for these devices.

closed 2022-10-14 by jascoproducts

deviantintegral, 2024-12-12:
@jascoproducts I believe I have discovered two bugs in your zigbee devices. One is more a performance issue, where the devices spam the network (reported at https://github.com/jascoproducts/firmware/issues/127).

The other is that after being powered on for several months without interruption, the devices stop responding to `Route Request` commands until they are restarted. This means they can't respond to commands from the coordinator. It would be great to see a firmware update fixing those!