jascoproducts/firmware #96: [Bug Report]: BRAND - ZW NUMBER - PRODUCT NUMBER - FW VERSION

bob-blake, 2022-10-14:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW3010

### Product Number

46203

### FW Version

5.54

### Bug Report Details

Very long latency between pressing the switch and the lights starting to turn on/off, regardless of min/max dim thresholds.  I don't have the means to time it precisely but it seems longer than 500ms.

I have some ZW4008/46201 non-dimmer switches in the same setup which used to have the same issue, but after updating their firmware from 5.53 to 5.54 the latency on those switches is almost fully eliminated.  Perhaps the same adjustment can be made to the ZW3010 dimmers.

This long latency also affects a user's ability to adjust the dimming by holding the up/down buttons - the lag between pressing and the dimming occurring makes it almost impossible to accurately set a brightness level.

### How To Reproduce

Press the "up" or "down" button and see how long it takes for the lights to respond.

bob-blake, 2022-12-13:
I'm going to push back on this a bit because it doesn't seem to have to do
with the LED turn-on/ramp behavior - it's a long delay between pressing the
switch and the start of the ramp.  You can easily see this delay if you
compare actuating the switch via a Z-Wave command vs. actuating it by
pressing the switch directly.  Turning on the lights happens nearly
instantaneously via Z-Wave, but pressing the switch has a very long latency
to turn-on.

jascoproducts, 2022-12-13:
Thanks for reaching back out. I apologize for the length of this response, but I want to provide some additional commentary and details about the dimmer.

There are 2 situations that will play into what you are describing.

- The first is there are no design standards for dimmable LED bulbs. To elaborate on what we described previously, some bulbs do not light immediately at 1% of the dimmer but instead require the dimmer to be at a higher level. In some extreme instances we have tested bulbs that wont light until they get closer to 10%. The dimmer is always ramping at a consistent rate and the higher % it takes to light the bulb will appear to be a long delay. This is ultimately what we were referencing in the previous response and is often the most common situation due to the enormous about of dimmable LED bulbs available on the market over the years.

With that said:

- If the bulbs you are using are not experiencing this. the second situation is the Dimmer has to have a delay when being manually turned on. Not all Z-Wave hubs/Controllers allow users to customize the features in the dimmer. For the most common features we offer button press methods to adjust/control the dimmer. For instance you can change the LED behavior, or manually reset the dimmer to factory defaults. Additionally, you have Double and Triple Tap functionality. Since the Dimmer has only two buttons, we must use a combination of these to perform configurations. The Dimmer has to be able to tell the difference between the customer trying to turn the light on and performing one of these operations.

We have reduced the delay as much as possible, and have landed on the current timing. the less of a delay you have, the faster you have to press the buttons and the harder it is to configure those features or perform double tap. This affects all users especially those who may have limited dexterity in their hands. We also tried updating logic to remove the delay assuming the user always wanted to turn it on to start, but that breaks the Double and Triple Tap functionality. If you wanted double tap or triple tap to control other devices only, you would be unable because as soon as you press the on button once it always turns on the local load as well. 

In some situations, the two could compound and appear that the dimmer has a much longer delay that it actually has. We are constantly looking to improve upon design and hardware with every new iteration as much as possible and this is definitely a point we look to improve with new models especially since this model is EOL and no longer manufactured.

Again, I apologize for the length, but hope we helped better explain our meaning from the first response.

Nick2253, 2022-12-28:
I'm also seeing this issue, and its especially pronounced in comparison to the older ZW3005/14294.  I have a hallway divided into two parts (one half with a 14294, and the other half with a 46203), with the exact same can light fixtures and exact same LED bulbs, and the on delay in quite obvious with the 46203.

I have the 46203 set to "Dim Quickly" using parameter 6, but compared to the control in the 14294, it just doesn't cut it.  On the 14294, I have parameters 9 and 10 set to 2 and 1 respectively.  However, if I change the 46203 to "Switch Mode" (parameter 16), it comes on instantly (but Switch Mode comes with its own problems, like disabling dimming at the paddle).  Switch Mode also seems to do nothing for Z-wave-controlled on/off.

It's possible that it's a LED threshold issue, but I'm not convinced.  I observe the delay on both the on and off curves.  When I manually adjust the brightness, I'm able to visually distinguish between about 90 and 100 brightness, and when I click it off, it doesn't look like it starts down the brightness curve until a sizeable delay.  I would agree with the ~500ms number as well.

Again, though, this delay does not appear to be there when the switch is in "Switch Mode".