jascoproducts/firmware #99: [Bug Report]: BRAND - ZW NUMBER - PRODUCT NUMBER - FW VERSION

yvessa, 2022-12-10:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZW1002

### Product Number

14297

### FW Version

5.51

### Bug Report Details

Uploading firmware 5.51 over 5.26 with ZWave JS seems to work without any error in the log. 

But version reported still 5.26 after uploading firmware 5.51.  

### How To Reproduce

Some people have the same issue than I have https://community.home-assistant.io/t/cant-upgrade-firmware-on-ge-enbrighten-zwave-plus-outlet-14297-zw1002/461356

Upload firmware 5.51 over 5.26 and look if version's field has really changed afterwards.

jascoproducts, 2022-12-13:
@yvessa Thanks for bringing this to our attention. We are looking into it now.

jascoproducts, 2022-12-20:
@yvessa since the 5.26 --> 5.51 update is not working for your 14297 it means that it did not ship with S2 support and will be unable to update to 5.51 (which is an S2 firmware). This is a special case where we did not change the model or the ZW number of the device in accordance with S2 implementation.

Closing this issue and pinning to discussion board.

closed 2022-12-20 by jascoproducts

bfarnham82, 2023-01-24:
If this outlet doesn't have S2 support, this should not be a video and feature listed on the amazon page. Why did I pay more than the white color to not get the feature?

https://amazon.com/gp/product/B08561Y4R3/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

jascoproducts, 2023-02-14:
Hey @bfarnham82,
 
Sorry about the confusion on this.
 
Jasco had released this device originally without S2/SmartStart with a small production run of Date Code 2022 or before (20 = Year 2020 and 22 = Week 22 / approximately May 31st). At that time, the listing and documentation reflected that it did not support S2/SmartStart. Once we updated to S2/SmartStart, all inventory was sold through and we updated to S2/SmartStart. Our listings were then updated to reflect the new features.

Unfortunately, Amazon does not keep 3rd party and 1st party inventory separate and will set and adjust pricing based on their own unique algorithms. If older inventory supplied by a 3rd party seller was shipped in place of ours, it is possible that you may have received older inventory. Amazon offers protection against cases where the product received doesn't match the listing and you could return it to Amazon for a refund or replacement.
 
If you are outside of Amazon's return policy and still have proof of purchase, you can contact our [Consumer Care team](https://byjasco.com//tech) with your info and they would be able to assist you in getting an S2 replacement. Let us know what you find out.

bfarnham82, 2023-04-03:
> Hey @bfarnham82,
>
> Sorry about the confusion on this.
>
> Jasco had released this device originally without S2/SmartStart with a small production run of Date Code 2022 or before (20 = Year 2020 and 22 = [...]

I have emailed your support, but never received a reply. can you assist with this?

noloarmi, 2023-04-18:
@jascoproducts 

Hello, this particular switch is on sale in Amazon for $19.  I bought 4 and all of them have the same upgrade issue.  I returned for and they send me 4 back, and all of them are still older stocks I guess.  So now, a total of 8 switches from Amazon are old stock.  I don't feel like doing a 3rd round of return/replace with Amazon.

Is there a chance to just contact support and I will get brand new .. newer almond models?

MollyrH, 2023-07-26:
I’m having the same issue with three ZW1002 almond outlets.  The boxes I have clearly indicate it supports S2 & SmartStart.  How do we know we can trust ordering new Jasco product since I’m assuming the new ones would say it supports the same.  Very difficult to check Date Codes on Amazon.

bfarnham82, 2023-07-26:
> > Hey @bfarnham82,
> > Sorry about the confusion on this.
> > Jasco had released this device originally without S2/SmartStart with a small production run of Date Code 2022 or before (20 = Year 2020 and 22 = Week 22 / approximately May 31st). At that time, the listing and documentation reflected that it did not support S2/SmartStart. Once we updated to S2/SmartStart, all inventory was sold through and we updated to S2/SmartStart. Our listings were then updated to reflect the new features.
> > Unfortunately, Amazon does not keep 3rd party and 1st party inventory separate and will set and adjust pricing based on their own unique algorithms. If older inventory supplied by a 3rd party seller was shipped in place of ours, it is possible that you may have received older inventory. Amazon offers protection against cases where the product received doesn't match the listing and you could return it to Amazon for a refund or replacement.
> > If you are outside of Amazon's return policy and still have proof of purchase, you can contact our [Consumer Care team](https://byjasco.com//tech) with your info and they would be able to assist you in getting an S2 replacement. Let us know what you find out.
> 
> I have emailed your support, but never received a reply. can you assist with this?
> 
> @jascoproducts

@jascoproducts any help here?

MollyrH, 2023-08-06:
I finally called the Consumer Care team number provided above - a very unpleasant discussion.  At first, after describing the issue of not being able to upgrade to the new firmware version, she had to "go talk with my manager".  Upon return, the agent said they had no record of any problems, and it was likely an issue with my hub (Home Assistant).  I told the agent I have dozens of other ZWAVE devices and this was the only one having this issue.  I also indicated it was document by the JASCO product support rep on GitHub and had assumed that was a JASCO employee.  The agent again had to "go talk with my manager".  The agent returned and asked me to send a link to the Github conversation which I did.  The agent showed it to the manager and then responded that this issue did not apply to my outlets (qty. 3 purchased from Amazon in mid-June 2023).  The agent asked if the box indicated the outlets supported S2 and SmartStart - which they do.  I replied that the date code is 2022 as was indicated above as the defective lot.  There is also no DSK code (the long series of numbers where you input the first 5 when adding a ZWave device) which the agent said had to be there for all S2 devices.  The agent had to "go talk with my manager" again, returned and said they would send me a return shipping label for only one outlet.  They indicated if they tested it and found it defective, they would send me a replacement.  I asked the agent about the other two outlets and was told if more than one outlet was not functioning properly then it was like an issue with my hub and I would need to resolve with hub manufacturer.  Clearly, even if they actually performed as they said they would, this was going to be a LONG process to resolve the three outlets.  Horrible customer support.

So, even though it was past the "1 month" return policy for Amazon, I called one of their reps.  After explaining that JASCO had provided faulty devices that did not meet the requirements on their package, the rep said they would absolutely provide support and take those units back.  She sent a UPS return label and I had a full refund 2 hours after dropping off at the local UPS Store - now THAT is customer support.  I used the refund to purchase a Leviton ZW15R, had it installed the next day and it works perfectly!  No more JASCO products for me....
[image]
[image]
[image]