jascoproducts/firmware #109: Unable to Update Firmware on ZW4002_55258-2

snel6424, 2023-01-31:
I am trying to update the firmware on my ZW4002_55258-2 fan control switch.  It is currently running firmware 5.50 (Original Release Firmware) and I am trying to update it to 0.51 (newest beta).  However, whenever I try to update the firmware, I get to about 40% and then get a timeout error.  The node then dies and I have to cut power to it, to get it to come back online.

I am using Home Assistant with the Z-Wave JS Addon, and a Zooz 700 stick.

I have factory reset the device twice and retried both times, but to no avail.

jascoproducts, 2023-12-04:
@snel6424 Were you ever able to get around this issue and update the device? If so, can you provide details?

snel6424, 2023-12-05:
> @snel6424 Were you ever able to get around this issue and update the device? If so, can you provide details?

No, I actually just tried again, and again it timed out.

jascoproducts, 2023-12-13:
@snel6424 Go ahead and contact our [Consumer Care Center](https://byjasco.com//tech) and they will get you taken care of.