jascoproducts/firmware #111: Feature requests for ZW3008 and ZW4006

ASoTNetworks, 2023-02-13:
Hello, not sure if you take feature requests but I have a bunch of ZW3008 and ZW4006 switches and wanted to see if any of these features can be added:
- parameter to disable local buttons to avoid people from turning switches on / off using the buttons
- parameter to use blue LED as motion status which would be useful for initial set up and debugging
- light sensor output to Z-Wave, if the sensor cannot output a lux value output 1 for bright and 0 for low light
- parameter to disable fade on / off when triggered by local buttons

anudub, 2023-02-23:
- parameter to disable local buttons to avoid people from turning switches on / off using the buttons

This is very useful with digitally controlled lights like ZigBee RGBW bulbs. The switch should report the button state i.e. upper button pressed or lower button pressed but not actually turn the relay on or off. The controller (Hubitat/SmartThings, etc.) can then digitally turn on/off the ZigBee lights based on the switch pressed. For comparison, this capability is available in Zooz switches, see - https://a.co/d/0HLrFJH. Zooz calls it the 'Smart Bulb' mode. Would be great if Jasco switches can support this capability too since most home automation controller based lighting is now done with ZigBee smart bulbs like Philips Hue, Sengled, etc. I am a Jasco fan but I desperately need this feature so I don't have to switch to Zooz switches as I have converted almost all the lighting in my house to ZigBee and I am tired of telling everyone to not turn off the switches. Thanks!

jascoproducts, 2023-02-24:
@ASoTNetworks We are looking into options for replacing the two models or redesigning them from the ground up because these two currently do not have additional memory available to update further beyond its current design. These two models were designed years ago using the older 500 series Z-Wave line. In order to re-certify any firmware, The Z-Wave Alliance requires the addition of S2 Security and SmartStart. This would actually cause us to remove features currently in it to accommodate the significant increase in code to add new features or even just to meet the Z-Wave Alliance requirements. While we cannot update the features of these models, we will take these under advisement for future models. Many of these suggestions have been requested in the past and make a lot of sense to provide additional versatility. 

@anudub Appreciate the additional information and context for your use case. As provided above, these 2 models specifically cannot be updated further due to memory constraints. We are adding this feature to new models in the future as we have started to see this request more and more. There is a potential workaround that will get you similar results. The Line and Load wire from the electrical box can be connected together on the same Line terminal on the back of the switch while leaving the load terminal empty. This will provide power to the switch and the bulb simultaneously regardless of the button press. While this doesn't allow you to change it back without rewiring, it could allow you to achieve the same result in some areas.

closed 2023-02-24 by jascoproducts