jascoproducts/firmware #112: How to update?

Skates1616, 2023-03-13:
I am using Z-Wave PC Controller and when I click update for a ZW4005 switch, it never starts, I only see this message and it never completes:

[image]

Any tricks I should be aware of?

jascoproducts, 2023-03-13:
Try these 2 options, let us know if any work:
Option 1: Remove and re-add the device and try again.
Option 2: Try using 39348's firmware.

Skates1616, 2023-03-14:
No go, I tried both options and am still receiving the same error.  I tried updating a Fan Switch (ZW4002) too and am receiving the same error.

I know updating works as I just updated a Inovelli switch as a test with the two firmware, it updated with no issue.

Any thoughts on what may be preventing the devices from updating?

jascoproducts, 2023-12-04:
@Skates1616 Did you ever manage to get this updated? If not, and you'd still be up to troubleshoot again, can you send us a screenshot of a Manufacturer_Specific_Get as shown below:

[image]

closed 2023-12-13 by jascoproducts