jascoproducts/firmware #118: [Bug Report]: Enbrighten - ZWA3016 - 58438 - 1.23.2

TheCranston, 2023-11-24:
### What brand is the bug report for?

Enbrighten-GE

### ZW Number

ZWA3016

### Product Number

58438

### FW Version

1.23.2

### Bug Report Details

Turning on the light, as the dimmer reaches full brightness there is a momentary flash off and on, and then the light is full on.  This flash at the end of the dim up cycle is visually unpleasant and hard on the eyes.  If I configure the dimmer into switch mode, the issue does not present itself.  But in dimmer mode, the default, it happens every time.  I've tried several brands of LED lights, including GE.  All of the bulbs exhibit the quick off and on flash right at the end of the turn on cycle.  I'm hoping there is a firmware update to resolve this issue.  thanks!

### How To Reproduce

turn the light on.  every single time.

jascoproducts, 2023-11-27:
@TheCranston Are you using a 3-way switch? If so, do you know if the wire out from the add-on switch (not the traveler wire from the dimmer itself) is connected to the load wire installed between the main dimmer and the fixture? This specific wiring configuration is not supported and is known to cause the issue you're describing.

In any case, we have a firmware file to solve your issue. We are currently preparing a release for several of our new devices and should have it out in the coming days. Stay tuned!

TheCranston, 2023-11-27:
@jascoproducts No.  Standard switching. No 3-way at all.  Looking forward to the new firmware.  Thanks!

jascoproducts, 2023-12-04:
@TheCranston see updated firmware [here](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZWA3016/58438%20-%20In-Wall%20Paddle%20Dimmer%2C%20700S). Let us know if the new firmware fixes your issue!

TheCranston, 2023-12-05:
@jascoproducts Thank you for the firmware update.  I've OTA updated the dimmer and it now reports the new version of 1.25.2.  However the issue of the light having a momentary flash while moving from 99% to 100% remains.  My scenario is this, the dimmer is in default dimming (not switch) mode.  The light is fully off.  Press the up paddle once.  The light begins its ramp up to full brightness.  At the threshold between 99% and 100% there is a momentary flash where the light appears to go out, then back on to 100% brightness.  This also happens when the light is commanded via z-wave to turn fully on.  If I tell the light to dim up to 99%, no flash.  Please let me know if there are a tests or debugs I can try to gather.  Thanks.

TheCranston, 2023-12-05:
Attached is a quick video showing the flicker.

https://github.com/jascoproducts/firmware/assets/3859141/789f44c7-10fb-4b9e-8019-e416fb25c062

jascoproducts, 2023-12-13:
@TheCranston Thanks for the vid. Could you switch the device to [1.24](https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZWA3016/58438%20-%20In-Wall%20Paddle%20Dimmer%2C%20700S/1.24) and let us know if that fixes it?

TheCranston, 2023-12-13:
@jascoproducts I downgraded to 1.24.2 as requested and the flash still happens at the 99% to 100% transition.

jascoproducts, 2023-12-13:
@TheCranston Is there any chance you could try another bulb with your updated firmware so we can absolutely rule that out as a coincidence?

TheCranston, 2023-12-13:
@jascoproducts sure thing.  I removed the Philips bulbs and put in GE relax led bulbs. no change.  The flash persists.

jascoproducts, 2023-12-13:
@TheCranston Can you try cutting off the top end of the threshold with Parameter 31, e.g. setting the value to a number less than 99 instead of 99 itself? You might lose a bit of max brightness but it may solve the issue for you. Let us know what happens.

<html xmlns:v="urn:schemas-microsoft-com:vml"
xmlns:o="urn:schemas-microsoft-com:office:office"
xmlns:x="urn:schemas-microsoft-com:office:excel"
xmlns="http://www.w3.org/TR/REC-html40">

<head>

<meta name=ProgId content=Excel.Sheet>
<meta name=Generator content="Microsoft Excel 15">
<link id=Main-File rel=Main-File
href="file:///C:/Users/nsmith/AppData/Local/Temp/msohtmlclip1/01/clip.htm">
<link rel=File-List
href="file:///C:/Users/nsmith/AppData/Local/Temp/msohtmlclip1/01/clip_filelist.xml">

</head>

<body link="#0563C1" vlink="#954F72">

Parameter   31 | Size 1 | Maximum   Brightness Threhold | Default Value 99 | Valid Values 1-99 | Press and hold   menu button for 5 seconds | Press 7 up and   1 down to begin cycling   between options
-- | -- | -- | -- | -- | -- | --

</body>

</html>

TheCranston, 2023-12-13:
@jascoproducts I decreased parameter 31 from its default of 99 until the flash stoped.  That was at 84.  incrementing back up to 85, and it flashes again.  That's a significant loss of brightness, so I'd not consider it a resolution, rather a temporary work around.  Hopefully the value of 84 points to something in the code that is useful.

ASoTNetworks, 2023-12-29:
I notice this same behavior with the motion dimmer 26933 / ZW3008 on firmware v5.32

brettcp, 2024-02-03:
@jascoproducts Same issue here with a ZW3010, flickers briefly when it ramps up to max brightness (same as in the video posted above).  If I set parameter 31 to 90, the flickering no longer occurs, but I'd rather not lose any brightness.. can this be fixed?  I'm running FW 5.55 on this switch.

Thanks!

TheCranston, 2024-02-19:
@jascoproducts are there any additional data or tests I can do to help in resolution of the flicker?  Thank you again for your support on this issue.

brettcp, 2024-02-19:
@jascoproducts  This issue still exists.. downgrading to 5.53 (as mentioned in my previous post) definitely helps and makes the flicker more tolerable, but its still there when the switch ramps up to full brightness.

TheCranston, 2024-06-17:
@jascoproducts greetings! Is there an update on the flicker issue which can be shared?  Please let me know if I can pull any traces or debug activities to help.  Thanks again.