jascoproducts/firmware #119: [Bug Report]: Jasco - ZW4008 - 46562 - 5.55 Double-tap up or down changes relay state. Only normal taps should change state.

nmpu, 2023-12-01:
### What brand is the bug report for?

Jasco

### ZW Number

ZW4008

### Product Number

46562

### FW Version

5.55

### Bug Report Details

How did this ever pass "certification"? Does anyone actually press the physical buttons to verify expected function?

v5.55 still reports central scene events (double-tap, triple-tap, and hold), BUT it also turns the load (relay) on and off when that should be limited to normal taps. I know exactly why you did it. The delay/debounce introduced to detect multiple taps can make the switch appear defective to someone expecting simple on/off. If you want to disable multiple taps, then add a configuration option.

### How To Reproduce

With v5.55 firmware, double-tap up or down. The relay should not change state, but it does. The original v5.53  works as expected.

jascoproducts, 2023-12-04:
Looking into this issue alongside #116

jascoproducts, 2023-12-13:
@nmpu In moving forward from 5.53, we originally did a bit of a cost-benefit analysis on this firmware for keeping the relay delay. For all those who enjoyed the function of the multi-tap, we had many more that complained about the delay, so we decided to remove it from future versions.

The only difference between 5.55 and 5.53 other than the delay removal is the adding of factory reset parameter 84. Please switch to 5.53 if you wish the multitap functionality.

closed 2023-12-13 by jascoproducts