jascoproducts/firmware #120: [Firmware Request]: GE - Plug-In Outdoor Switch, 300S

jamenlang, 2023-12-03:
### What brand is the request for?

Enbrighten-GE

### ZW Number

ZW4201

### Product Number

screenshot

jamenlang, 2023-12-03:
[image]

I know these things belong in a museum, but it's not my fault they work so well.

Just wanted to see if there's an updated firmware from FW: v1.24

jascoproducts, 2023-12-04:
@jamenlang Happy to hear they are still alive and kicking!
The technology of the Z-Wave module at the time of the 300 series was not designed to accept OTA updates. There is no updated firmware available for this device; you are already set up with the latest version 👍