jascoproducts/firmware #122: GE ZW4002 14287/55258

SurgeHawk, 2023-12-07:
I have 4 GE ZW4002 fan controllers. All were purchased at different times and new/used because it's hard to find them new anymore.

How can I tell which product number I have? They all say ZW4002. I want to update the firmware, but I don't want to put the wrong one on.

The latest firmware listed for each product:
14287 - firmware version 5.24
55258 - firmware version 5.50

jascoproducts, 2023-12-13:
@SurgeHawk They are both indeed ZW4002 but have different manufacturer-specific coding, so they might not be cross-compatible with each other's firmware. That's why you are seeing different availability for both versions; we have the Z-

The two versions you are seeing are correct; those are both the latest release firmwares for their respective devices.

SurgeHawk, 2023-12-15:
> @SurgeHawk They are both indeed ZW4002 but have different manufacturer-specific coding, so they might not be cross-compatible with each other's firmware. That's why you are seeing different availability for both versions.
> 
> The two versions you are seeing are correct; those are both the latest release firmwares for their respective devices.

I understand that they are correct version. 

I would like to know how to identify the 2 versions. I do not see any identifying information other then ZW4002

nmpu, 2024-01-03:
The **55258** has product ID **0x3337**. The **14287** has product ID **0x3131**. You should be able to find those numbers in your software. Otherwise, if you look at the _back of the switch_, you'll find the actual model number.