jascoproducts/firmware #124: [Firmware Request]: UltraPro - ZWA3016 - 59350-2

glenne, 2024-02-19:
### What brand is the request for?

UltraPro

### ZW Number

ZWA3016

### Product Number

59350-2

glenne, 2024-02-19:
I currently have version 1.23.  I tried the ZWA3016_UltraPro_59350_1.25.gbl file using Hubitat but it said the 'The transferred image does not match the Firmware Target'

jascoproducts, 2024-02-19:
Thanks. We're looking into this.

jascoproducts, 2024-02-28:
@glenne Just replaced the existed v1.25 with a newly coded version specific to 59350. Let us know if this solves your issue.

closed 2024-02-28 by jascoproducts

heatvent, 2024-06-16:
Hi, I was looking to update my GE Enbrighten ZWA3016 dimmers with the firmware from here:

https://github.com/jascoproducts/firmware/tree/main/zwave/Enbrighten-GE/ZWA3016/58438%20-%20In-Wall%20Paddle%20Dimmer%2C%20700S/1.25

There is reference to this issue (#124). However, this seems to be an UltraPro firmware rather than a GE Enbrighten firmware. When I try to update with Hubitat to a GE ZWA3016 it says it does not match the target.