jascoproducts/firmware #127: Zigbee devices load network with route request broadcasts

deviantintegral, 2024-06-14:
Hello! First off, I want to say how happy I am with the Enbrighten Zigbee switches I have. They're great devices that everyone, including guests are happy to use.

I see this repository is still focused on Z-Wave, not Zigbee devices. https://github.com/jascoproducts/firmware/issues/95 noted that there's no known reports of issues with the Zigbee firmware, so there's no reason to release updates. Even if the @jascoproducts team here is only working on Z-Wave, I'm hoping you can forward this on to the team responsible for the zigbee products.

I have 10 various Enbrighten switches (and 2 plugs) such as https://www.zigbee2mqtt.io/devices/43076.html#enbrighten-43076 on my network. While investigating an unrelated issue, I noticed that all of my Enbrighten devices are sending `Route Request` broadcasts every 60 seconds:

[image]

To confirm this wasn't an issue with my coordinator or Zigbee software, I created a new network with a Conbee II running deCONZ with only a single switch connected. It showed exactly the same behaviour with route requests every 60 seconds.

As is, this is problematic for people like myself who have purchased many Enbrighten products. A typical zigbee network can handle one broadcast a second. On my own, it takes about 3 seconds for all of the broadcast retransmits to settle. This leads to additional latency when other commands are trying to get sent during the broadcasts.

I don't see this behaviour with any other devices, or at least not enough to be noticeable. It would be great if this could be fixed, and even better if it was done so in a way ZHA and zigbee2mqtt could use.

jasondefuria, 2024-07-02:
Yes, @jascoproducts I had asked about a firmware update before. I didn't do the troubleshooting that @deviantintegral did until tonight. I too see that the Enbrighten Zigbee Swithces are "Chatty" in requesting routes. Depending on how often they communicate, this causes rate limiting, where messages are then delayed in sending via the Zigbee mesh.

tdaddy-evie, 2024-09-08:
Third for asking for updated firmware. Fixing this issue would also be a great time to add double/triple and long press actions as well as Relay Control options to the Zigbee line of products to bring it up to par with the Z-Wave line, given that Z-Wave is on its way to being deprecated in favor of Zigbee/Thread at this point.

deviantintegral, 2024-12-12:
Hey folks, I just filed a docs PR in zigbee2mqtt documenting a bug with multiple Enbrighten devices that you might be interested in. They stop responding to route requests, which means they don't respond to device commands from the coordinator. Honestly, it's not a problem if you have regular power outages. But if you're on reliable power, they can have uninterrupted power for months at a time and start acting up.

While these devices are so nice to install, and physically well constructed, these two issues make it hard to recommend using them. One or two switches, and you probably wouldn't notice any issues. I installed 11 of them, which means I'm seeing an issue on at least one device every month or so, which is getting frustrating for the rest of the family!