jascoproducts/firmware #130: [Bug Report]: UltraPro - ZWA4011 - 59347 - 1.22.2

rrendec, 2024-10-29:
### What brand is the bug report for?

UltraPro

### ZW Number

ZWA4011

### Product Number

59347

### FW Version

1.22.2

### Bug Report Details

There is a visible delay between the moment when the rocker switch is pressed and the moment when the relay toggles. Ideally there should be no delay. Most people would expect the lights to turn on the moment they press the button.

I am assuming the delay is there because without it it's impossible to differentiate between a single click and a double click, which correspond to different actions.

However, I also own a few 39348 devices, which used to have the same problem, and it was fixed by upgrading to the latest firmware, v5.54. With this firmware, they are still able to differentiate between a single click and a long press (hold) and act as a scene controller, but with no delay between the click and the relay control. From my point of view, this is the ideal behavior, and still allows me to implement more advanced controls (e.g. using something like [Z-Wave JS UI](https://github.com/zwave-js/zwave-js-ui) and [Node-RED](https://nodered.org/)). I would buy more 39348 devices, but unfortunately they are no longer available.

Is there any chance you could implement something similar for the 59347 devices? I understand this would break the existing double click function, but the user could choose which firmware to install depending on the behavior they prefer (immediate relay control or double click support). Or even better, the behavior could be selected by a configuration parameter, and then the user would be able to choose the desired behavior very easily.

### How To Reproduce

Press the rocker switch once (single click) to toggle the current relay (load) state. The issue is always reproducible.

rrendec, 2024-10-31:
This could be much easier than I thought initially. It could be enough to expose the double click timeout value as a user configurable parameter over Z-Wave e.g. through the Configuration Command Class. By "double click timeout" I mean the amount of time the device waits for a second click before it identifies the user action as a single click.

I'm thinking that the double click logic could remain unchanged, and if the user wants a faster response on single click and doesn't care about double click, they can just configure the timeout to be shorter.

tech165, 2025-05-06:
Hi, this is still not fixed and a lot of users complains of this issue on my side. Is there any advancements in a new firmware ? Since the fix was already implemented in the 500S version, it should be an easy fix. This is a downgrade from the old switch. Thanks

rrendec, 2025-10-31:
I agree that having different firmware versions is not a great way to fix it. Which is why I suggested adding a parameter to make it configurable, as a better solution. I also agree that functionality should not be removed in newer versions. The parameter can either enable/disable double tap completely or configure the double tap timeout. The configurable timeout option is simpler/cleaner, and people like me could set it to a low value like 10ms, which is unnoticeable to the eye. But either way, it would make both parties happy.

And, just like with any software, when a new configuration parameter is added, the default value should be consistent with the behavior of previous versions, where the parameter did not exist. Otherwise, it would create frustration/confusion for people who upgrade.