jascoproducts/firmware #132: [Bug Report]: UltraPro - ZWA3016 - 59350 - 1.25

Yura80, 2025-03-19:
### What brand is the bug report for?

UltraPro

### ZW Number

ZWA3016

### Product Number

59350

### FW Version

1.25.2

### Bug Report Details

If the device is set to switch mode, it's not possible to switch it off by a Zwave command. When switch mode is off, everything is working as expected.
I tried different firmware versions (1.23 and 1.25), re-interview, exclude and re-include.

I'm using it with Home Assistant 2025.1.2. No issues with any other switches/dimmers.

### How To Reproduce

1. Set parameter 16 Switch Mode to 1. All other parameters set to default.
2. Switch device OFF in Home Assistant.
3. Device stays ON

jascoproducts, 2025-03-19:
@Yura80 

We weren't able to reproduce this on the PC Controller. Is there a way to send strings of data via HA? Try sending `26 01 00 00` (multilevel set - off) and `26 01 FF 00` (multilevel set - on).

[image]

Yura80, 2025-03-19:
Hi! Thank you so much for the quick response. I believe this is the command HA sets.
Please see this video: https://photos.app.goo.gl/hTtBPnMxERsAfXiU8

Looks like it's being set 0, then immediately restores back to 99, but the light stays on, doesn't even flicker.

I will try to get more detailed logs from HA.

Yura80, 2025-03-19:
Log from Z-Wave JS UI while trying to switch off. It looks like it confirms the command, but then reports that the level is still 99:

```
2025-03-19T17:41:17.530Z DRIVER » [Node 097] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      78
                                  └─[SupervisionCCGet]
                                    │ session id:      13
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 0
2025-03-19T17:41:17.567Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -85 dBm
                                  └─[SupervisionCCReport]
                                      session id:          13
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s
2025-03-19T17:41:18.845Z DRIVER « [Node 097] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -85 dBm
                                  └─[MultilevelSwitchCCReport]
                                      current value: 99
                                      target value:  0
                                      duration:      0s
```

Yura80, 2025-03-19:
I've also tried another brand new dimmer switch of the same type, firmware 1.23.2, and the issue is 100% reproducible.

Yura80, 2025-03-19:
Another log with raw command data:

```
2025-03-19T19:01:32.977Z DRIVER   one or more queues busy
2025-03-19T19:01:32.983Z SERIAL » 0x011600a900010062086c019304260100ff25000000008aa6                  (24 bytes)
2025-03-19T19:01:32.985Z DRIVER » [Node 098] [REQ] [SendDataBridge]
                                  │ source node id:   1
                                  │ transmit options: 0x25
                                  │ callback id:      138
                                  └─[SupervisionCCGet]
                                    │ session id:      19
                                    │ request updates: true
                                    └─[MultilevelSwitchCCSet]
                                        target value: 0
2025-03-19T19:01:32.991Z SERIAL « [ACK]                                                                   (0x06)
2025-03-19T19:01:32.995Z SERIAL « 0x010401a90152                                                       (6 bytes)
2025-03-19T19:01:32.999Z SERIAL » [ACK]                                                                   (0x06)
2025-03-19T19:01:33.002Z DRIVER « [RES] [SendDataBridge]
                                    was sent: true
2025-03-19T19:01:33.058Z SERIAL « 0x011d00a98a000006029ea8b87f7f0101034a070000020100007f7f7f7f7f79    (31 bytes)
2025-03-19T19:01:33.060Z SERIAL » [ACK]                                                                   (0x06)
2025-03-19T19:01:33.064Z DRIVER « [REQ] [SendDataBridge]
                                    callback id:            138
                                    transmit status:        OK, took 60 ms
                                    repeater node IDs:      74, 7
                                    routing attempts:       1
                                    protocol & route speed: Z-Wave, 40 kbit/s
                                    routing scheme:         LWR
                                    ACK RSSI:               -98 dBm
                                    ACK RSSI on repeaters:  -88 dBm, -72 dBm
                                    ACK channel no.:        1
                                    TX channel no.:         1
2025-03-19T19:01:33.179Z SERIAL « 0x011300a80000010062056c0213ff00009e007f7f3e                        (21 bytes)
2025-03-19T19:01:33.181Z SERIAL » [ACK]                                                                   (0x06)
2025-03-19T19:01:33.183Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -98 dBm
                                  └─[SupervisionCCReport]
                                      session id:          19
                                      more updates follow: false
                                      status:              Success
                                      duration:            0s
2025-03-19T19:01:33.186Z CNTRLR   [Node 098] [~] [Multilevel Switch] currentValue: 99 => 0          [Endpoint 0]
2025-03-19T19:01:33.189Z DRIVER   all queues idle
2025-03-19T19:01:34.332Z SERIAL « 0x011300a80000010062052603630000009e007f7ffa                        (21 bytes)
2025-03-19T19:01:34.337Z SERIAL » [ACK]                                                                   (0x06)
2025-03-19T19:01:34.342Z CNTRLR   [Node 098] [~] [Multilevel Switch] currentValue: 0 => 99          [Endpoint 0]
2025-03-19T19:01:34.347Z CNTRLR   [Node 098] [~] [Multilevel Switch] duration: {"value":0,"unit":"s [Endpoint 0]
                                  econds"} => {"value":0,"unit":"seconds"}
2025-03-19T19:01:34.352Z CNTRLR   [Node 098] [~] [Multilevel Switch] targetValue: 0 => 0            [Endpoint 0]
2025-03-19T19:01:34.356Z DRIVER « [Node 098] [REQ] [BridgeApplicationCommand]
                                  │ RSSI: -98 dBm
                                  └─[MultilevelSwitchCCReport]
                                      current value: 99
                                      target value:  0
                                      duration:      0s
```

Yura80, 2025-03-19:
Found a solution. It works if "Transition Duration" is set to "0s" (it's empty by default).

[image]

jascoproducts, 2025-03-19:
Glad you found a solution! Thanks for letting us know.

closed 2025-03-19 by jascoproducts

Yura80, 2025-03-19:
@jascoproducts I still feel like this is a firmware bug. 26 01 00 FF means "default transition duration" and should be handled properly.

Yura80, 2025-03-20:
Unfortunately, the workaround I suggested only works when switching the device from the Zwave JS settings page.

If controlled from the main HA UI, it still sends 26 01 00 FF which should be properly handled by the device.

bobcat0070, 2025-08-15:
@Yura80 were you able to resolve this in any way?
@jascoproducts can we please reopen this issue?